sudomesh/LoRaLayer2 #3: LoRaLayer2 shouldn't refer to "WebSockets"

tlrobinson, 2020-01-21:
It seems like this library shouldn't specifically refer to "WebSockets" even thought that's how `disaster-radio` currently uses it, i.e. this change: https://github.com/sudomesh/LoRaLayer2/commit/2f6ecd1b0f8d97aae9d8bc964a12b9f452d18c7f although I agree `_chatBuffer` wasn't a great name either.

I think a more generic name like `_messageBuffer` would be appropriate.

paidforby, 2020-01-21:
Agreed. chatBuffer was a bad name. That's why I changed it. wsBuffer isn't an ideal name either. I think I considered messageBuffer, but I have so many local variables called "message". Really, I just need to go through and decide what parts are called what. Like when is piece of data a packet and when is it a message. And what buffer handles what data. Of course, it should be as generic as possible, since websocket is not the only option for an upper-layer.

paidforby, 2020-02-12:
Latest commit address this issue. I've renamed the buffers to be more descriptive,
`outBuffer` is now `L1OutBuffer`
`inBuffer` is now `L1InBuffer`
`wsBuffer` is now `L3OutBuffer`

There is still no `L3InBuffer`, because I have yet to come across as need, though I could imagine it being useful if there are multiple users on the same node.

If you this is an adequate solution to your issue, feel free to close.

closed 2020-03-11 by paidforby