sudomesh/LoRaLayer2 #7: Move _localAddress/_hashTable from Layer1 to LoRaLayer2?

tlrobinson, 2020-02-05:
I'm curious why `Layer1` has things like `_localAddress` and `_hashTable`. It seems like they aren't used internally, only by `LoRaLayer2`. Would it make sense to move them to `LoRaLayer2`, especially since [`Layer1` is considered the physical layer](https://github.com/sudomesh/disaster-radio/wiki/Layered-Model)?

AFAICT the only difference between them in `Layer1_LoRa` and `Layer1_Sim` is the latter has part of `isHashNew` commented out for some reason: https://github.com/sudomesh/LoRaLayer2/blob/master/src/Layer1_Sim.cpp#L88-L93

paidforby, 2020-02-12:
I took your advice for localAddress, it does make sense for that to be part of LL2. The hashing is not currently being used. When hashing is needed, I imagine it will need to be heavily refactored, so I've left it where it is. 

The reason these were in the Layer1 interface code may have had something to do with the simulator. I know the simulator has to use a different hashing library and that the local address is set by a the javascript server. When I eventually revisit the simulator, I'm sure I can figure out sensible solution to sync with these changes.

paidforby, 2020-04-20:
Recent updates to LoRa Layer 2 remove hashing completely, see https://github.com/sudomesh/LoRaLayer2/commit/cf1e9f8921c8802ad5e8706534b919e64199c09d. More thought needs to be put in to how to use hashing effectively. There is some discussion of using hashes to handle the rebroadcast of messages, here https://github.com/sudomesh/disaster-radio/issues/55. Further discussion could be held there, or a new issue could be opened.

closed 2020-04-20 by paidforby