sudomesh/LoRaLayer2 #9: Rename to "LoRa Mesh Layer2"

samuk, 2020-02-12:
A bunch of people will search 'lora mesh' every day. 

Almost none will search 'loralayer2. Is it worth renaming for visibility / SEO?

tlrobinson, 2020-02-12:
I’d just call it LoraMesh but there’s a project named that already https://github.com/nootropicdesign/lora-mesh

paidforby, 2020-02-22:
TL:DR
@samuk you could make a "LoRa Mesh" website that then directs visitors to this repo. But I'm not interested in renaming this repo with "mesh" somewhere in the name. You can read my reasoning below.

I shy away from the term "mesh." Despite it's scientific definition, it comes with a degree of baggage. What is a mesh? The answer is subjective and opinionated. There are community networks that call themselves meshes, but are really star topology networks that are inspired by meshing more in their organizational structure than in their network design. There are corporations and for-profit companies that have co-opted the term and tack it on to whatever device they are selling because it talks to another identical device using a proprietary protocol. All of them are within their rights to use the term because there is no standard that describes what is and is not a mesh.

While I like to think that what we are designing is more of a true mesh network than those projects, I know that the implementation does not always match the design.

I'd prefer our project to more closely associated with the OSI model and the standards derived from it (ethernet, MAC, TCP/IP, BGP, UDP). What we are designing is not another mesh, it is another standard for communication. It is another internet, a smaller, minimal, supplemental internet, but still an inter-network.

All this is not to discourage anyone from publicizing this project as "LoRa Mesh" or "LoRa Mesh Layer 2" if you think it would be beneficial. Right now, I'm not concerned with publicizing the project, I'm concerned with getting it working. Perhaps the two are related, but I don't have the time to work on both.

samuk, 2020-02-22:
Fair enough :)

I don't have the skills to make it work, but I'm happy to help publicise it.. Maybe down the line it would be worth doing the domain thing.

closed 2020-02-22 by samuk