sudomesh/LoRaLayer2 #14: Example Program failed to compile

mamanbudiman, 2020-06-07:
I try to compile the example program,
but it is failed to compile.

Also, I have seen that the source library is different with
the previous one.
Could you update the example program for the new library ?

Regards,

paidforby, 2020-06-08:
Sorry about that. You are correct, I have not synced the example with the latest changes to the source. Really, the platformio.ini should point to a specific commit (or, ideally, a tagged release) of the code. I'll quickly fix this by adding a commit tag to the [lib_deps line](https://github.com/sudomesh/LoRaLayer2/blob/master/examples/platformio.ini#L20), I think 3f650076c1e1ff97d33c8aab20fb844528822542 was the last one that worked with the the API in the example.

The real solution of updating the API, may take a little more time for me to get to.

paidforby, 2020-06-08:
That latest commit now builds for me. Let me know if you have any more problems. I'll close this issue once I fix this problem correctly by updating the API in the example (and the README while I'm at it).

mamanbudiman, 2020-06-08:
Ok, thanks you. I will check it.

Regards,

paidforby, 2020-06-28:
Finally got around to properly and fully resolving this issue. Example is now updated to use the latest version of the LL2 library. Should be working as of  15e30154f32edf12db7a375c2db48699e06674fc, I tagged that commit as 1.0.1 also.

closed 2020-06-28 by paidforby