sudomesh/LoRaLayer2 #15: Example compilation fail for  SAMD21 in Arduino IDE

bobricius, 2020-07-08:
Hi, I am beginner in this high level programing, 
I try compile this library in Arduino IDE for Armachat project with SAMD21 (etc Arduino Zero)
I get lot of errors and I have no idea how to fix.
Including errors in file.
[error.txt](https://github.com/sudomesh/LoRaLayer2/files/4892525/error.txt)
Can anybody help?
THANKS

paidforby, 2020-07-08:
The most likely problem is that you need to `#define ARDUINO_LORA` as described here https://github.com/sudomesh/LoRaLayer2#api. 

If you are using the example code, you may be able to place `#define ARDUINO_LORA` at the very top of the main.ino file. This is meant to be done in the [`platformio.ini` file](https://github.com/sudomesh/LoRaLayer2/blob/master/examples/platformio.ini#L21) using a build flag. I'm not sure if it is possible to add a build flag in the Arduino IDE.

paidforby, 2020-07-08:
It looks like there might be a way to add compiler flags in  a `boards.txt` or `platform.txt` file. The extra flag you would need to add is `-DARDUINO_LORA`. Sorry I can't be of more help, I don't use the Arduino IDE much these days. I highly recommend trying out platformIO, I bet they support your board. A quick search produced these results, https://docs.platformio.org/en/latest/platforms/atmelsam.html and https://docs.platformio.org/en/latest/boards/atmelsam/sparkfun_samd21_mini_usb.html

bobricius, 2020-07-09:
Thanks for reply, I already add #define ARDUINO_LORA but nothing changed
also memcpy are udeclared, I try add
#include <stdio.h>
#include <string.h>
anywhere but no change :(

paidforby, 2020-07-09:
Those memcpy's and memcmp's are throwing errors because the correct header is not being included which is happening because ARDUINO_LORA is not being defined at a global level (i.e. at compile time). 

I guess a solution could be to make ARDUINO_LORA the default by adding a guard at the top of Layer1_LoRa.h and LoRaLayer2.h that would define ARDUINO_LORA as long as the other two options (RL_SX1276 and SIM) are not defined. I'll give this a shot and let you know if it works.

paidforby, 2020-07-09:
That solution doesn't appear to work for me in platformio, it causes linker errors now. I don't think there is a good work around to not using a build flag. By creating a default setting, I can now compile, but I am not able to link. I'm not sure why the linker errors are happening, but I'm guessing it doesn't like ARDUINO_LORA not being globally defined.

bobricius, 2020-07-09:
I included
#define ARDUINO_LORA 
in to LoRaLayer2.h and Layer1_LoRa.h
and errors are reduced only to:
`sketch\router_beacon.ino.cpp.o: In function `setup':

D:\arduino-1.8.8-d11\portable\sketchbook\router_beacon/router_beacon.ino:23: undefined reference to `Layer1Class::Layer1Class()'

D:\arduino-1.8.8-d11\portable\sketchbook\router_beacon/router_beacon.ino:24: undefined reference to `Layer1Class::setPins(int, int, int)'

D:\arduino-1.8.8-d11\portable\sketchbook\router_beacon/router_beacon.ino:25: undefined reference to `Layer1Class::setLoRaFrequency(unsigned long)'

D:\arduino-1.8.8-d11\portable\sketchbook\router_beacon/router_beacon.ino:26: undefined reference to `Layer1Class::init()'

libraries\LoRaLayer2\LoRaLayer2.cpp.o: In function `LL2Class::daemon()':

D:\arduino-1.8.8-d11\portable\sketchbook\libraries\LoRaLayer2\src/LoRaLayer2.cpp:573: undefined reference to `Layer1Class::getTime()'

D:\arduino-1.8.8-d11\portable\sketchbook\libraries\LoRaLayer2\src/LoRaLayer2.cpp:581: undefined reference to `Layer1Class::getTime()'

D:\arduino-1.8.8-d11\portable\sketchbook\libraries\LoRaLayer2\src/LoRaLayer2.cpp:582: undefined reference to `Layer1Class::transmit()'

D:\arduino-1.8.8-d11\portable\sketchbook\libraries\LoRaLayer2\src/LoRaLayer2.cpp:587: undefined reference to `Layer1Class::getTime()'

D:\arduino-1.8.8-d11\portable\sketchbook\libraries\LoRaLayer2\src/LoRaLayer2.cpp:588: undefined reference to `Layer1Class::spreadingFactor()'

D:\arduino-1.8.8-d11\portable\sketchbook\libraries\LoRaLayer2\src/LoRaLayer2.cpp:597: undefined reference to `Layer1Class::receive()'

D:\arduino-1.8.8-d11\portable\sketchbook\libraries\LoRaLayer2\src/LoRaLayer2.cpp:577: undefined reference to `Layer1Class::getTime()'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board MattairTech MT-D21E (rev B).`

bobricius, 2020-07-09:
I am using this LoRa Library https://github.com/sandeepmistry/arduino-LoRa

paidforby, 2020-07-10:
Yes, those are the linker errors I was talking about.

Ok, so I've found the solution. You need to put the compiler flag in a file named `platform.local.txt` in the same directory where `platform.txt` is stored. For me, this looks like, `~/Downloads/arduino-1.8.13-linux64/arduino-1.8.13/hardware/arduino/avr` because I'm on Ubuntu just running arduino from the folder where I downloaded it, so it may look different for you. 

In the `platform.local.txt` file, you need to put `compiler.cpp.extra_flags=-DARDUINO_LORA`. That will get rid of all the errors except one about a `Serial.printf` in `Layer1_LoRa` on line 155, you can comment that line out and it should compile. Idk why `Serial.printf` would not be working, that is "standard" Arduino, that needs to be investigate into further.

For reference, I found that solution here, https://stackoverflow.com/questions/28037491/arduino-1-0-6-how-to-change-compiler-flag

bobricius, 2020-07-15:
same erros as above 
D:\arduino-1.8.8-d11\portable\sketchbook\router_beacon/router_beacon.ino:23: undefined reference to `Layer1Class::Layer1Class()'
..... and more
much less than without #define ARDUINO_LORA in .h files