sudomesh/LoRaLayer2 #16: Compile failure from master

robgil, 2020-09-10:
Similar to https://github.com/sudomesh/LoRaLayer2/issues/15, I'm having compile issues. Opening as a separate issue as I'm trying from master with the fixes introduced in #15. 

## Code is based on the example as follows.
```
#include <Arduino.h>
#include <Layer1_LoRa.h>
#include <LoRaLayer2.h>

#define ARDUINO_LORA

//#define OLED_SDA 4
//#define OLED_SCL 15
//#define OLED_RST 16

#define LORA_CS 18
#define LORA_RST 23
#define LORA_IRQ 26
#define LORA_FREQ 915E6

Layer1Class *Layer1;
LL2Class *LL2;

void setup()
{
  Serial.begin(115200);
  delay(200);

  Serial.println("* Initializing LoRa...");

  Layer1 = new Layer1Class();
  Layer1->setPins(LORA_CS, LORA_RST, LORA_IRQ);
  Layer1->setLoRaFrequency(LORA_FREQ);
  if (Layer1->init())
  {
    Serial.println(" --> LoRa initialized");
    LL2 = new LL2Class(Layer1); // initialize Layer2
    LL2->setLocalAddress("c0d3f00d"); // this should either be randomized or set using the wifi mac address
    LL2->setInterval(10000); // set to zero to disable routing packets
  }
  else
  {
    Serial.println(" --> Failed to initialize LoRa");
  }
}

void loop()
{    
  LL2->daemon();
}
```
Using a Heltec ESP32 LoRa V2 board.

## Errors
```
pio run
Processing heltec_wifi_lora_32_V2 (platform: espressif32; board: heltec_wifi_lora_32_V2; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/heltec_wifi_lora_32_V2.html
PLATFORM: Espressif 32 (2.0.0) > Heltec WiFi LoRa 32 (V2)
HARDWARE: ESP32 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing sudomesh/LoRaLayer2#72c67c3c7fcc23dbcb6daf6d6e6ef1ed2372f430
Library Manager: LoRaLayer2 @ 1.0.1 has been installed!
Library Manager: Installing dependencies...
Library Manager: LoRa @ 0.7.2 is already installed
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LoRaLayer2> 1.0.1
|   |-- <LoRa> 0.7.2
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
Building in release mode
Linking .pio/build/heltec_wifi_lora_32_V2/firmware.elf
.pio/build/heltec_wifi_lora_32_V2/libbcc/libLoRaLayer2.a(LoRaLayer2.cpp.o):(.literal._ZN8LL2Class6daemonEv+0x8): undefined reference to `Layer1Class::getTime()'
.pio/build/heltec_wifi_lora_32_V2/libbcc/libLoRaLayer2.a(LoRaLayer2.cpp.o):(.literal._ZN8LL2Class6daemonEv+0xc): undefined reference to `Layer1Class::transmit()'
.pio/build/heltec_wifi_lora_32_V2/libbcc/libLoRaLayer2.a(LoRaLayer2.cpp.o):(.literal._ZN8LL2Class6daemonEv+0x10): undefined reference to `Layer1Class::spreadingFactor()'
.pio/build/heltec_wifi_lora_32_V2/libbcc/libLoRaLayer2.a(LoRaLayer2.cpp.o):(.literal._ZN8LL2Class6daemonEv+0x1c): undefined reference to `Layer1Class::receive()'
.pio/build/heltec_wifi_lora_32_V2/libbcc/libLoRaLayer2.a(LoRaLayer2.cpp.o): In function `LL2Class::daemon()':
/home/rgil/Documents/PlatformIO/Projects/test1/lib/LoRaLayer2/src/LoRaLayer2.cpp:36: undefined reference to `Layer1Class::getTime()'
/home/rgil/Documents/PlatformIO/Projects/test1/lib/LoRaLayer2/src/LoRaLayer2.cpp:36: undefined reference to `Layer1Class::getTime()'
/home/rgil/Documents/PlatformIO/Projects/test1/lib/LoRaLayer2/src/LoRaLayer2.cpp:36: undefined reference to `Layer1Class::transmit()'
/home/rgil/Documents/PlatformIO/Projects/test1/lib/LoRaLayer2/src/LoRaLayer2.cpp:36: undefined reference to `Layer1Class::getTime()'
/home/rgil/Documents/PlatformIO/Projects/test1/lib/LoRaLayer2/src/LoRaLayer2.cpp:36: undefined reference to `Layer1Class::spreadingFactor()'
/home/rgil/Documents/PlatformIO/Projects/test1/lib/LoRaLayer2/src/LoRaLayer2.cpp:36: undefined reference to `Layer1Class::receive()'
/home/rgil/Documents/PlatformIO/Projects/test1/lib/LoRaLayer2/src/LoRaLayer2.cpp:36: undefined reference to `Layer1Class::getTime()'
.pio/build/heltec_wifi_lora_32_V2/src/main.cpp.o:(.literal._Z5setupv+0x34): undefined reference to `Layer1Class::Layer1Class()'
.pio/build/heltec_wifi_lora_32_V2/src/main.cpp.o:(.literal._Z5setupv+0x38): undefined reference to `Layer1Class::setPins(int, int, int)'
.pio/build/heltec_wifi_lora_32_V2/src/main.cpp.o:(.literal._Z5setupv+0x3c): undefined reference to `Layer1Class::setLoRaFrequency(unsigned int)'
.pio/build/heltec_wifi_lora_32_V2/src/main.cpp.o:(.literal._Z5setupv+0x40): undefined reference to `Layer1Class::init()'
.pio/build/heltec_wifi_lora_32_V2/src/main.cpp.o: In function `setup()':
/home/rgil/Documents/PlatformIO/Projects/test1/src/main.cpp:26: undefined reference to `Layer1Class::Layer1Class()'
/home/rgil/Documents/PlatformIO/Projects/test1/src/main.cpp:27: undefined reference to `Layer1Class::setPins(int, int, int)'
/home/rgil/Documents/PlatformIO/Projects/test1/src/main.cpp:28: undefined reference to `Layer1Class::setLoRaFrequency(unsigned int)'
/home/rgil/Documents/PlatformIO/Projects/test1/src/main.cpp:29: undefined reference to `Layer1Class::init()'
collect2: error: ld returned 1 exit status
*** [.pio/build/heltec_wifi_lora_32_V2/firmware.elf] Error 1
===================================================== [FAILED] Took 3.09 seconds =====================================================
```

_platformio.ini_
```
$ cat platformio.ini 
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:heltec_wifi_lora_32_V2]
platform = espressif32
board = heltec_wifi_lora_32_V2
framework = arduino
lib_deps = 
	sudomesh/LoRaLayer2#72c67c3c7fcc23dbcb6daf6d6e6ef1ed2372f430
```

Interested in getting this working as I'd like to contribute to the project and add additional features.

robgil, 2020-09-14:
For platformio, I added the build flag as follows and it was able to compile.

[image]

closed 2020-09-14 by robgil

paidforby, 2020-09-14:
Thanks for opening (and closing) this issue. Yes, you definitely need the `ARDUINO_LORA` build flag it is included in the example platformio.ini, https://github.com/sudomesh/LoRaLayer2/blob/master/examples/platformio.ini#L21 and is kinda of mentioned in the README here, https://github.com/sudomesh/LoRaLayer2#api