sudomesh/LoRaLayer2 #19: Can't link when two copies of library exist

paidforby, 2020-09-20:
One small issue generated by recent pull request #18. It seems to have broken my ability to link using platformio when I have a copy of LoRaLayer2 in my `lib` directory, but I am building using the copy made by platformio.ini in the  `.pio` directory. This is a very minor issue and only has a small impact on my dev cycle. It should go unnoticed by anyone who isn't actively developing the LL2 library.

The error generated is as follows,
```
Linking .pio/build/ttgo-lora32-tbeam/firmware.elf
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/sender/src/main.cpp.o:(.bss.Layer1+0x0): multiple definition of `Layer1'
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/receiver/src/main.cpp.o:(.bss.Layer1+0x0): first defined here
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/sender/src/main.cpp.o:(.bss.LL2+0x0): multiple definition of `LL2'
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/receiver/src/main.cpp.o:(.bss.LL2+0x0): first defined here
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/sender/src/main.cpp.o:(.data.MAC+0x0): multiple definition of `MAC'
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/receiver/src/main.cpp.o:(.data.MAC+0x0): first defined here
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/sender/src/main.cpp.o: In function `setup()':
main.cpp:(.text._Z5setupv+0x0): multiple definition of `setup()'
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/receiver/src/main.cpp.o:main.cpp:(.text._Z5setupv+0x0): first defined here
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/sender/src/main.cpp.o:(.bss.counter+0x0): multiple definition of `counter'
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/receiver/src/main.cpp.o:(.bss.counter+0x0): first defined here
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/sender/src/main.cpp.o:(.data.GATEWAY+0x0): multiple definition of `GATEWAY'
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/receiver/src/main.cpp.o:(.data.GATEWAY+0x0): first defined here
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/sender/src/main.cpp.o: In function `loop()':
main.cpp:(.text._Z4loopv+0x0): multiple definition of `loop()'
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/receiver/src/main.cpp.o:main.cpp:(.text._Z4loopv+0x0): first defined here
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/sender/src/main.cpp.o:(.data.LOCAL_ADDRESS+0x0): multiple definition of `LOCAL_ADDRESS'
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/receiver/src/main.cpp.o:(.data.LOCAL_ADDRESS+0x0): first defined here
.pio/build/ttgo-lora32-tbeam/src/main.ino.cpp.o:(.bss.LL2+0x0): multiple definition of `LL2'
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/receiver/src/main.cpp.o:(.bss.LL2+0x0): first defined here
.pio/build/ttgo-lora32-tbeam/src/main.ino.cpp.o: In function `loop()':
main.ino.cpp:(.text._Z4loopv+0x0): multiple definition of `loop()'
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/receiver/src/main.cpp.o:main.cpp:(.text._Z4loopv+0x0): first defined here
.pio/build/ttgo-lora32-tbeam/src/main.ino.cpp.o: In function `setup()':
main.ino.cpp:(.text._Z5setupv+0x0): multiple definition of `setup()'
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/receiver/src/main.cpp.o:main.cpp:(.text._Z5setupv+0x0): first defined here
.pio/build/ttgo-lora32-tbeam/src/main.ino.cpp.o:(.bss.Layer1+0x0): multiple definition of `Layer1'
.pio/build/ttgo-lora32-tbeam/src/lib/LoRaLayer2/examples/receiver/src/main.cpp.o:(.bss.Layer1+0x0): first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/ttgo-lora32-tbeam/firmware.elf] Error 1
```

It looks like is doesn't like the mulitple platformio projects in the examples directory. I'm not sure how to fix this or if I even care enough to fix it, but I figured I'd post it here in case someone comes across this problem and is more inclined to solve it.

paidforby, 2020-09-20:
I actually just remembered the solution, and the actual cause of this problem. @robgil used `main.cpp` in both the receiver and sender examples, this is not allowed when compiling a local copy of a library in arduino. It has something to do with the fact that the arduino framework creates its own main.cpp, I opened an issue about this same problem with the build system makeESParduino some months ago, https://github.com/plerup/makeEspArduino/issues/134. Still wish there was just a way to ignore that examples directory, but I haven't found a straightforward way to do that with platformio.

closed 2020-09-20 by paidforby

robgil, 2020-09-20:
> Closed #19 <https://github.com/sudomesh/LoRaLayer2/issues/19> via efaa3fa

Oooo ok. That makes sense. I can update the samples.

paidforby, 2020-09-20:
I think I already took care of the problem. I don't think your platformio.ini files will care that I renamed the main files to `main.ino`, though, admittedly, I have yet to test your example code.

commit efaa3fa7 (2020-09-20): rename main.cpp files to main.ino, fixes #19